christianhaitian/arkos #311: There was an error with installing this update

ajsaucier, 2021-04-03:
On an RG351P, on the previous version. Went to update to the latest one and it says "There was an error with installing this update". Is this is a known problem or could it be specific to my device?

christianhaitian, 2021-04-03:
Try enabling remote services so time in your unit can be synced.

ajsaucier, 2021-04-03:
Perfect, that worked. You're amazing, man. Thank you.

closed 2021-04-03 by ajsaucier

ashencos, 2021-04-17:
how do i update on rg351p to version 1.06 since i always find 1.05? i don't have wifi

christianhaitian, 2021-04-17:
You wait for a new image.  A final one will be available by the end of this month.

ashencos, 2021-04-17:
can you also insert the Italian language in the menu?

ashencos, 2021-04-17:
or if you tell me where to download the files to insert other languages thank you

christianhaitian, 2021-04-17:
What italian language?  All available languages for the ES used in ArkOS is already included.

ashencos, 2021-04-17:
there is no english in the languages ​​you can add?

ashencos, 2021-04-17:
there is no Italian only English in the languages ​​can you add?

christianhaitian, 2021-04-17:
All available languages are already included. EN, BR, DE, ES, FT, KO, PT, ZH-CN.  That's it unless someone else creates additional ones to be added.  https://github.com/christianhaitian/EmulationStation-fcamod/tree/master/resources/locale

ashencos, 2021-04-17:
only Italian is missing can you put on?

christianhaitian, 2021-04-17:
If you or someone else know how to do the translation from English to Italian and create the translation file to do so, sure.

ashencos, 2021-04-19:
i noticed that the games go slow is there any setting to put on any emu?

christianhaitian, 2021-04-19:
Please open a new ticket for new issues and provide more specific information about your issue.

ashencos, 2021-04-22:
where should the pc engine roms be inserted? I created the folder but does not see them and also the pcfx ones? thank you

christianhaitian, 2021-04-22:
See this section of the wiki here: https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information