christianhaitian/arkos #313: Fix for shaders no longer automatically loading with the latest ArkOS 1.6 (04/03/2021) update

christianhaitian, 2021-04-03:
To fix shaders no longer auto loading in retroarch and retroarch32 with the latest ArkOS 1.6 update:

1.  Load retroarch or retroarch32.
2.  Then go to settings>video>auto-shader delay and set that to 0.  
3.  Save the configuration.

Now your shaders should automatically load with your retroarch emulators when configured to do so.

closed 2021-04-03 by christianhaitian

loganir, 2021-04-16:
The is still a problem. Now you can save the shaders, but it only saves one effect of the shaders
Exemple
If I load and Save `lcd-grid-v2-gba-color-motionblur.glslp ` it will load 3 differents effects, the GBA LCD Grid (v2), the GBA Color pallet correction and the motionblur, but if I save, clode the retroarch and open de the again it will have ONLY the GBA LCD Grid (v2) loaded, the other ones will disapear from the shaders loaded list.

loganir, 2021-04-19:
Now that I'm reading I can see that my phone auto correct changed a few words.
Just instaled a clean img on my SD from the version 1.5 that was working with `settings>video>auto-shader delay 3` then loaded `lcd-grid-v2-gba-color-motionblur.glslp` and was working fine. Then updated to the last version on this weekend and it changed the delay to 0 and saved the 3 effects on the shaders for my GBA, but after reboot (the retroarch) it does not load, but it is there, just need to enter on the menu and click on apply changes. BUT on the other emulators that I use different shaders (I think it is only one) it loads as usual. I will try to test something else later and post more feedback here.