christianhaitian/arkos #316: ogage blink and return to current brightness.

southoz, 2021-04-04:
Hey Christian! 

When using ogage I noticed there is no user feedback on setting Perfnorm or Perfmax. Extended on the unused blink routines you wrote, the below reads in the current brightness value from the output of brightnessctrl, and blinks once for off and twice for on:

```
use std::process::{Command, Stdio};

fn blinkon() {

    let output = Command::new("brightnessctl").arg("g").stdout(Stdio::piped()).output().expect("Failed to execute brightnessctl");
    let current = String::from_utf8(output.stdout).unwrap();
    Command::new("brightnessctl").args(&["s","0"]).output().expect("Failed to execute brightnessctl");
    Command::new("sleep").arg("0.2").output().expect("Failed to Sleep");
    Command::new("brightnessctl").args(&["s","160"]).output().expect("Failed to execute brightnessctl");
    Command::new("sleep").arg("0.2").output().expect("Failed to Sleep");
    Command::new("brightnessctl").args(&["s","0"]).output().expect("Failed to execute brightnessctl");
    Command::new("sleep").arg("0.2").output().expect("Failed to Sleep");
    Command::new("brightnessctl").arg("s").arg(current).output().expect("Failed to execute brightnessctl");
}

fn blinkoff() {

    let output = Command::new("brightnessctl").arg("g").stdout(Stdio::piped()).output().expect("Failed to execute brightnessctl");
    let current = String::from_utf8(output.stdout).unwrap();
    Command::new("brightnessctl").args(&["s","0"]).output().expect("Failed to execute brightnessctl");
    Command::new("sleep").arg("0.3").output().expect("Failed to Sleep");
    Command::new("brightnessctl").arg("s").arg(current).output().expect("Failed to execute brightnessctl");
}
```

The functions are then called during actions with no user feedback:

```
        else if ev.event_code == PERF_MAX {
            Command::new("perfmax").arg("none").output().expect("Failed to execute performance");
            blinkon();
        }
        else if ev.event_code == PERF_NORM {
            Command::new("perfnorm").arg("none").output().expect("Failed to execute performance");
            blinkoff();
        else if ev.event_code == WIFI_ON {
            blinkon();
            Command::new("nmcli").args(&["radio","wifi","on"]).output().expect("Failed to execute wifi on");
        }
        else if ev.event_code == WIFI_OFF {
            Command::new("nmcli").args(&["radio","wifi","off"]).output().expect("Failed to execute wifi off");
            blinkoff();
        }
```

christianhaitian, 2021-04-05:
Cool.  Thanks for the code.  Out of curiosity, why do you include function for a wifi on and wifi off since the ogs doesn’t have builtin wifi?

southoz, 2021-04-05:
Some users have a wifi dongle in the USB A port and they do halve the battery life. It can be disabled through the Wifi Menu but it's nice to have a hotkey.

closed 2021-04-12 by christianhaitian