christianhaitian/arkos #317: sd card bricked

levidunc, 2021-04-06:
my sd card will no longer allow me to add any games or manage files like i cant even format it

christianhaitian, 2021-04-06:
Sounds like your sd card is either write protected or has gone into read only mode due to reaching max write capabilities. What brand sd card is it and if using it in a sd card adapter in your computer, does the adapter have a sd card write lock that is activated that you can deactivate?

levidunc, 2021-04-08:
the original

levidunc, 2021-04-08:
where would the write lock be

christianhaitian, 2021-04-08:
The problem is the fact you used the original sd card that came with the unit.  Those are known to fail quickly as they are not good quality.  Per the wiki, get a good authentic name brand card such as Sandisk or Samsung.

closed 2021-04-08 by christianhaitian