christianhaitian/arkos #318: End of the line for ArkOS on RG351P

deadkat37, 2021-04-07:
I read your 4/4/2021 announcement regarding ArkOS on the RG351P with great sadness as your OS has made this unassuming device fly beyond my wildest expectations.

With a heavy heart I would like to express my huge respect & gratitude for the work you have put into the system and hope that Anbernic take your concerns about the internal USB controller SOC decisions seriously as I think you have personally made a massive contribution to the success of the device. 

Many thanks and all the best for the future.

joseMartinezDev, 2021-04-07:
I would like to say thanks as well, your work has been really impressive, actually, I bought my RG351P because of ArkOS. So, THANK YOU SO MUCH, CHRISTIAN! and good luck with your future projects.

christianhaitian, 2021-04-08:
Thank you both for the kind words.  I hope ArkOS will continue to bring you enjoy for sometime.  :)  Take care.

closed 2021-04-08 by christianhaitian

EmilyLove26, 2021-04-13:
I missed the announcement, where can I read it?

joseMartinezDev, 2021-04-13:
Here is the announcement
https://github.com/christianhaitian/arkos/wiki/Announcement