christianhaitian/arkos #321: Unable to Save Shaders

32gits, 2021-04-09:
### Describe the Issue (If applicable)
Unable to save shader config in retroarch between sessions. If I load a shader (lcd3x for example) and save it, it does not persist and remain active on next boot of retroarch.

### How can the issue be reproduced? (If applicable)
Load a shader preset in retroarch.
Save shader preset for content directory or core or game
and/or
Save override for content directory or core or game
Exit retroarch
Boot retroarch and load content. Shader is not active.

### What device are you using?
RG351M

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
1.6 (04032021)

### Anything else you'd like to include that may help to review this issue or feature request?

icaroferre, 2021-04-09:
Also experiencing this same issue. Running 1.6 on an RG351P.

christianhaitian, 2021-04-09:
Fix is here: https://github.com/christianhaitian/arkos/issues/313

closed 2021-04-09 by christianhaitian