christianhaitian/arkos #323: TIC 80 Savegame load crashing

fry-guy, 2021-04-11:
The savegame function for Tic 80 is not working with this game: https://tic80.com/play?cart=873

The game description contains some details related to this issue. But it suggests it should work as long as you're not trying to play in the Chrome browser. (It says "if you are playing this seriously, consider downloading the cart and playing in the TIC-80 program.")

Could this issue be looked into? I'm not certain if it's affecting other games with save function, but I suspect it might.

Additionally, RetroArch save states also do not seem to work with the Tic 80 core, although you do not get any message saying the feature is not supported as you do with other cores that do not support savestates. Perhaps this is also an error and not an unsupported feature?

Thank you!

christianhaitian, 2021-04-11:
I'll try updating the libretro core to see if these issues have been addressed.  https://github.com/libretro/TIC-80

christianhaitian, 2021-04-13:
Hi,

Can you help test this updated tic-80 core and see if it resolves this issue?

1. Take the linked compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, launch a Tic-80 game.

Please confirm if it resolves the issue for you or not.  

Thanks,
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6301274/arkosbackup.tar.gz)

fry-guy, 2021-04-13:
I can try this later tonight. However, I think I will need to flash a spare SD card for the test. I don't want to wipe all of my other settings, and it looks like "Restore Settings" resets all of your EmulationStation collections and several standalone emulator configurations? https://github.com/christianhaitian/arkos/issues/37#issuecomment-735480872

I've got some large collections, among other things, so sounds like this would undo a lot of work if I ran it on my main card?

christianhaitian, 2021-04-13:
This won't.  It will only replace the existing tic80 core.  Nothing wrong with precaution though.

christianhaitian, 2021-04-14:
Greetings, 

Any update on this?

Thanks,

fry-guy, 2021-04-17:
Hey, thanks for checking in. I wasn't able to get to this until tonight, unfortunately.

So, the good news is this does appear to have resolved the saving issue. I can exit out and resume my save without crashing.

The bad news is the game has become a lot slower with bad audio stutters.

I tested out a couple other games (Mario Bros., Witch 'em Up), and they aren't as noticeably slow. But the game with the saving issues is now not really playable, mostly due to the glitchy audio.

Not sure if there's some settings or anything else that can be done about it? I checked the game options in RetroArch but there aren't very many for this core.

christianhaitian, 2021-04-17:
sounds like there was a regression that caused that for the game.  Were there any updates to the cart that may help?

fry-guy, 2021-04-17:
Hmm.. I don't think so. It looks like the last update was 2019-06-26? http://tic80.com/play?cart=873

Github page for the game also hasn't been updated in 2 years: https://github.com/btco/so12l

I'm not certain what you mean by a regression?

christianhaitian, 2021-04-17:
Meaning something got worse with the emulator instead of improve.

fry-guy, 2021-04-17:
Ah, I see.

Hey, I just realized the SD I tested the updated emulator on was still running ArkOS 1.5, whereas the card where I originally detected this issue is on 1.6.

Might that be making any difference in the speed?

christianhaitian, 2021-04-17:
Roll back of 1.6 to 1.5 is only possible by flashing an old image.  If you want to roll back to the previous tic-80emulator, just update the core through retroarch.

fry-guy, 2021-04-17:
That's good to know. Thanks!

I went ahead and updated my spare card to the latest version of ArkOS, and I see it's still running the newer version of TIC-80 (V0.90.1579-dev). Still got the slowdown issue, unfortunately.

fry-guy, 2021-04-18:
Hey, where do you pull the latest version of this core? The file size doesn't seem to match any of the TIC-80 Linux cores at https://buildbot.libretro.com/nightly/

christianhaitian, 2021-04-18:
I build it directly from source from this link https://github.com/libretro/TIC-80

christianhaitian, 2021-04-27:
The issue seems to be related with a compatibility between that specific game and this libretro tic-80 emulator.  Closing as this is not something I can resolve myself and should be reported upstream on nesbox's TIC-80 github [here](https://github.com/nesbox/TIC-80/issues) for further review and advisement.

closed 2021-04-27 by christianhaitian