christianhaitian/arkos #326: Mame 2003 cores load with a black screen

dabbler77, 2021-04-13:
When a mame 2003 core is installed to retroarch (64 or 32bit) and a rom associated with that core, when you launch that rom a black screen is displayed. Pressing the hot key (select+x) brings up the retroach menu, and when you resume, the game screen is correct.

To replicate:
In retrorarch 64or32, download Mame 2003 xtreme (or any mame 2003 core)
Edit /etc/emulationstation/es_systems.cfg and add the new core into the correct section for arcade
Edit the rom metadata, select retroarch and the mame 2003 core
Launch the rom
Black screen
Press hotkey
Select resume
Screen is normal.

Rom tested here is rtypeneo.zip located in \roms\arcade

dabbler77, 2021-04-14:
After further testing, this is caused only when RetroArch on screen notifications are turned off. Really odd. Turn them on, all roms load perfectly. Turn on screen notifications off and *some* roms load with a black screen.

christianhaitian, 2021-04-14:
Interesting.  That mean it's a retroarch issue that should be reported upstream.  Thanks for reporting this for the record.

closed 2021-04-14 by christianhaitian