christianhaitian/arkos #327: Drastic not saving in-game or save states

jessman1988, 2021-04-15:
Hello.  I'm running ArkOS 1.6 on my RG351M. I've noticed that Drastic is saving my configuration options ie; frameskip: none, but not saving my progress in games ; Kingdom Hearts Re-Coded or Super Mario 64 DS. I've tried creating save states and those also disappear upon returning to the emulator. 

I'm using a fresh 256GB Samsung Evo Memory Card and scanned to fix the exact partition. I've also noticed that randomly the scraped art and information seems to vanish from the main interface. Currently my dream cast images refuse to show up but the videos will still play.  

Any assistance would be appreciated. Thank you for the incredible firmware.

christianhaitian, 2021-04-16:
It sounds like you may have deleted some important folders from the NDS folder in the easyroms partition.  Restore them by extracting them from this [archive](https://github.com/christianhaitian/arkos/raw/main/defaultromsfolderstructure-rg351p.tar)
For the disappearing content, it sounds like you may be suffering some corruption of the exfat roms partition.  This will happen if you frequently just reboot the unit using the reset button instead of doing a safe shutdown or restart.  Try scanning the easyroms partition from a Windows 10 or newer computer to see if it can fix it.

jessman1988, 2021-04-17:
Thank you very much for the prompt response.  I'm used to devices like Raspberry Pis so I take great care when safely shutting down these kinds of devices.  I can't seem to download the archive from the link.  I expanded the exfat partition from a 128gb back-up image. Perhaps that may have caused the issue?  I'll likely start over, backing up my roms, images and videos and see if the issue persists.

christianhaitian, 2021-04-17:
Sorry about that.  Try this link. https://github.com/christianhaitian/arkos/raw/main/defaultromsfolderstructure-rg351p.7z

closed 2021-04-20 by christianhaitian

jessman1988, 2021-04-29:
I realized the issue was due to expanding the roms partition after flashing a 128gb backup.  i started fresh from a clean install and the issue is completely resolved.  hope this wasn't a big inconvenience.  thank you so much for your help. <3