christianhaitian/arkos #329: Problem saving shadder presets (per core, folder, etc)

sebvalmontx360, 2021-04-17:
### How can the issue be reproduced? (If applicable)

It is easy to reproduce the problem. If I change retroarch settings inside any game and I choose "core override" or "folder override", everything saves, except shadders. I would like to use shadder hq2x with my 2D games (SNES, Genesis, etc) but I have to load it manually every time I open a game. I also tried to save the shadder as "folder preset" and "core preset", but it didn't work also. I also tried with RGA scalling enabled and disabled, but it didn't do the magic also. I can save shadder presets in my other retroarch based devices (NVidia Shield and PC).

### What device are you using?
RG351M

### What version of ArkOS are you on?
ArkOS 1.6

christianhaitian, 2021-04-17:
Have updated to today’s update yet?

sebvalmontx360, 2021-04-17:
I did an update 8 hours before and it was not correct. But I just did a new update and I can confirm that now shader presets are loading automatically. Thank you very much,

closed 2021-04-17 by sebvalmontx360