christianhaitian/arkos #330: ppssppgo for 351v

haveanair1, 2021-04-18:
First of all, thank you for making the test beta.
The ppssppgo for 351v has good performance, so I added it to the es-system and ran it. The screen comes out flat. Can the resolution be used for 351v?

christianhaitian, 2021-04-18:
I'll see if I can figure this out.  I was thinking ppssppgo wouldn't work on the V.  I'll test it and see what I figure out.  Stay tuned.

haveanair1, 2021-04-22:
Thank you. If you can run ppssppgp at a resolution worthy of 351v, it will be a truly dreamlike sight. I'll wait. Take good care of me.

evan, 2021-06-30:
It looks like this is the commit we need: https://github.com/OtherCrashOverride/ppsspp-go2/commit/b67ecd41621a0b040efb0a6805a37b6c4e730d06

Not sure where to get a prebuilt binary.

christianhaitian, 2021-06-30:
I tried that already and the screen is still off unfortunately.

evan, 2021-06-30:
Ah ok. I assume you tried hardcoding different values for the rg351v?

I extracted the latest build from the OGA firmware (https://wiki.odroid.com/odroid_go_advance/os_image/ubuntu_es#v22) and it runs, but the screen is vertically squished. Is that the same problem you had? 

Input doesn't seem to work but I don't know how to fix that, so I can't get to the menus.

[PPSSPPSDL.zip](https://github.com/christianhaitian/arkos/files/6742154/PPSSPPSDL.zip)

christianhaitian, 2021-07-01:
Correct.

evan, 2021-07-01:
Ok. I will try to get a build environment set up on the device. Or is it better to cross-compile?

Do you know what I need to do to get input to work?

christianhaitian, 2021-07-01:
For a build environment, you can setup a chroot in a VM.  Here’s one that I use.  [https://forum.odroid.com/viewtopic.php?p=306185#p306185](https://forum.odroid.com/viewtopic.php?p=306185#p306185)
For controls, you’ll need to revert this commit [https://github.com/OtherCrashOverride/ppsspp-go2/commit/5d50e3bcd75ad170ce945b1d74526739980bfbfd](https://github.com/OtherCrashOverride/ppsspp-go2/commit/5d50e3bcd75ad170ce945b1d74526739980bfbfd) so you can make use of sdl based controls.

evan, 2021-07-02:
Perfect, ok. Do you build PPSSPP for 32bit or 64?

christianhaitian, 2021-07-02:
64

evan, 2021-07-02:
I got the build environment working fine and can ship new builds to the device.

Do you recommend starting from your `ppsspp-go2` repo and trying to roll forward, or starting from https://github.com/OtherCrashOverride/ppsspp-go2 and trying to revert the inappropriate changes?

christianhaitian, 2021-07-02:
Honestly, try both and see which one works out best.

evan, 2021-07-08:
Starting from upstream seems closest; the game renders more or less correctly but it seems like the framebuffer isn't being written to the window at the correct vertical scale; it is compressed. This is a surprise to me because the vertical resolution of the odroid go super is also 480, so I expected the error to be in the other direction.

The menu is rendered in the wrong place as well, but differently wrong, and it seems like the fonts are corrupt.

I will keep messing around with it.

christianhaitian, 2021-07-08:
Sounds like some similar issues I ran into.