christianhaitian/arkos #333: 351p Blank screen after new artwork on game startup

AdamtheAtom, 2021-04-20:
Once I select a game and run it, after the new gameboy art logo pops up, the sound of the game plays, but nothing shows on the screen unless I bring up retroarch, then the game shows as normal.  Not sure if something I did is causing it but would love to have it addressed if possible.  I have only tested gba games, but every single one does it.  Thanks!

christianhaitian, 2021-04-20:
When did you start experiencing this issue?

AdamtheAtom, 2021-04-21:
I just updated to the latest version last night, so since then.

christianhaitian, 2021-04-21:
What version of ArkOS were you on before you updated last night?  Trying to pinpoint which update may have caused this for you.

christianhaitian, 2021-04-21:
Also are you using shaders?

AdamtheAtom, 2021-04-21:
Unfortunately I can't remember the exact update.  Probably an update from a few weeks ago, as I haven't had the 351 more than a month.  I updated maybe twice in that time, and then of course last night.  I also don't use any shaders.  Not sure if the gameboy logo after you start a game is the culprit, but it's the only thing that seems different from before that may have led to no video playing.

markyparky4286, 2021-04-21:
Same here... I have narrowed it down to Retroarch32, any emulator using Retroarch32 has a blank screen when loading a game, only sound is present, but seems to be ok for emulators that use Retroarch.

tomatoskittles, 2021-04-23:
I have a slightly different variation of this issue.. For me only GBA has no video (audio plays). All other emulators are working And the other difference: even pressing select-X to bring up the RetroArch does not show anything. All I can do is Select-Start twice to exit.

Only starting happening after updating today.

I forget which version I had before, but I last updated on March 12th.

If I recall correctly, GBA may be the only emulator that I did NOT set a shader on. Not sure if that helps.

tomatoskittles, 2021-04-23:
Update: I tried changing the emulator for GBA:

No video for: MGBA and MGBA_RUMBLE
All other emulators show video.

tomatoskittles, 2021-04-23:
Update 2:

I logged into the terminal and removed this folder:

/home/ark/.config/retroarch/config/mGBA

Which I assume contained core overrides to the retroarch configuration.

Now MGBA and MGBA_RUMBLE work with video again.

closed 2021-04-27 by christianhaitian