christianhaitian/arkos #335: Daphne does not work on roms2/ 2nd SD card on RG351V

terryterryd, 2021-04-21:
When attempting to get Daphne to work, I eventually got it working by putting the files in the roms directory (as opposed to roms2).  I have specified that ArkOS uses the SD card, but this emulator seems to ignore that.

christianhaitian, 2021-04-21:
Have you updated the ArkOS install since you installed the image?

terryterryd, 2021-04-21:
Running OaD 1.6 (I updated today).  Thanks

christianhaitian, 2021-04-21:
They'll be an update that will correct the script that caused that.  When the update is available and you've completed it, just switch to main, then switch back to sd2 and it should correct the issue for you.

terryterryd, 2021-04-21:
Thanks... You aren't the only firmware that takes a bit of Daphne-manipulation.  The "other 2 firmwares" were configurable from their EmulationStation es_systems scripts.  I could not find an equivalent for ArkOS, hence a ticket.  Cheers.

christianhaitian, 2021-04-22:
New update for ArkOS OaD Test Version:

-Added UAE4arm_libretro.so for retroarch32 for Amiga and Amiga32
-Added potator core for Watara Supervision
-Added section for MD-MSU  using the genesis_plus_gx retroarch core
-Updated /usr/bin/emulationstation.sh and switch to sd2 and switch to main scripts to be more stable.
-Fixed Daphne not loading from SD2
-Updated emulationstation to support scraping for Watara Supervision

Just update from the Options section

closed 2021-04-22 by christianhaitian