christianhaitian/arkos #338: RG351V suggestions to improve

agb2, 2021-04-23:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on? Test release 1.7
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?
1. PSP text particularly on "Jeanne de Arc" loses its format and I have added a file found online to the opt\ppsspp font folder which improves the text and is supposedly the official psp font. It really works well and I have also added the font to my other device RG351P. Can you consider adding this please.
2. Can you give the option through PPSSPP to change the screen resolution to fully fit the 4:3 screen on the RG351V. I am aware that this will distort the screen slightly, but I would be happy to have the option. The official RG351V firmware provided with the device already allows for PSP games to be played in 4:3

christianhaitian, 2021-04-23:
Official PSP fonts can not be legally be included. I’ll checkout the 4:3 option.

christianhaitian, 2021-04-24:
New update for ArkOS OaD Test Version:

-Added ppsspp-stock emulator as default (Hopefully helps with performance)
  * Savestates are not interchangeable with the regular ppsspp in ArkOS OaD.
-Added ability to restore retroarch and retroarch32 default settings from Options/Advanced menu

Just update from the Options section