christianhaitian/arkos #339: RG351V: No more systems shown after Update to Test Release 1.8

Gordon1972, 2021-04-24:
Hi,
i just updated to ArkOS Test Release 1.8 (from 1.6) via Wifi on my RG351V.
After the Update was finished, the search for the systems took 1 second an no systems from my 50+ systems
are shown. ArkOS ist just showing "RetroArch" and "Options" in the main menu.
I use a single 128GB SanDisk SD-Card with the EASYROMS-Partition in exFAT format.

Could you please investigate what went wrong? The "V" is now rendered useless...

Thanks for looking into it..

christianhaitian, 2021-04-24:
Ooops...my bad.  Forgot to put a check in to see if a second sd card is being used and adjust the updated es_systems.cfg based on that or not.  I've added an update for that to fix it.  Just update as usual and you should be at 1.8.1.  Thanks for reporting this.

Gordon1972, 2021-04-24:
Thank you for fixing it so quickly.

closed 2021-04-24 by christianhaitian