christianhaitian/arkos #342: Do not abandon the development of the Rg351p

fabinhojcr, 2021-04-24:
Your operating system has become very mature and good for rg351p, please keep developing for it.

christianhaitian, 2021-04-24:
Further development for the P/M will not continue.  As you stated, it's very mature for the platform as is and will serve as a very good platform for emulation for PS1 titles and lower and some PSP, NDS, N64 and Dreamcast games are fairly playable depending on preference.  Thanks for the kind words.

closed 2021-04-24 by christianhaitian

fabinhojcr, 2021-04-24:
Will it be possible to update the cores of the retroarch when one is available?

christianhaitian, 2021-04-24:
Yes.  Core updates will continue to function as long as the repo it's set to continues to get updates which I and a few others are planning to keep updated as needed.