christianhaitian/arkos #343: PSX don't rumble

elpeina, 2021-04-24:
PSX games do not rumble, missing "device type: retropad with analog" option is required.  In ELEC yes rumble

christianhaitian, 2021-04-24:
Did you make sure to use the default pcsc_rearmed rumble core?  If so, did you follow these directions for enabling rumble? 
https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-enable-rumblevibration-in-pscx_rearmed

elpeina, 2021-04-24:
> Did you make sure to use the default pcsc_rearmed rumble core? If so, did you follow these directions for enabling rumble?
> https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-enable-rumblevibration-in-pscx_rearmed

Don't work

christianhaitian, 2021-04-24:
This is not 351Elec.  The setup is different for this with ArkOS.  I went through the instructions I linked in the previous email and confirmed rumble worked fine using Mr. Driller.  Reread the instruction carefully and follow that.  Don't look for a feature from another system.  Follow the instructions given here carefully.  Also make sure you did not change the default emulator from retroarch32 with pcsx_rearmed_rumble.

elpeina, 2021-04-24:
> This is not 351Elec. The setup is different for this with ArkOS. I went through the instructions I linked in the previous email and confirmed rumble worked fine using Mr. Driller. Reread the [...]

Not work. I have done everything in the manual

christianhaitian, 2021-04-25:
what game are you testing with?

elpeina, 2021-04-25:
> what game are you testing with?

A lot of games

christianhaitian, 2021-04-25:
Sorry I can't help any more on this issue.  I've tested psx rumble following the directions provided in the wiki and it works fine for me.  I've asked for what game are you testing and you've given me no example.  Maybe try a fresh image and if that doesn't work, go back to 351Elec.

closed 2021-04-25 by christianhaitian

elpeina, 2021-04-25:
> Sorry I can't help any more on this issue. I've tested psx rumble following the directions provided in the wiki and it works fine for me. I've asked for what game are you testing and you've given me no example. Maybe try a fresh image and if that doesn't work, go back to 351Elec.

Thanks for the help

christianhaitian, 2021-04-25:
As I stated, when I used Mr. Driller following the directions, it vibrates fine for me.  I just had to also make sure vibration was turned on within the game's option menu as well.

elpeina, 2021-04-25:
> As I stated, when I used Mr. Driller following the directions, it vibrates fine for me. I just had to also make sure vibration was turned on within the game's option menu as well.

does not work with vibration activated in game.

I use a custom Arkos (last versión), I will try original arkos

christianhaitian, 2021-04-25:
What custom ArkOS?  Where did you get it from?

elpeina, 2021-04-25:
> What custom ArkOS? Where did you get it from?

BobArkos, I'm going to install it again

fabinhojcr, 2021-04-25:
@elpeina How good that someone will continue the development, where do I find it?

christianhaitian, 2021-04-25:
I know nothing about BobArkOS.  For support for that OS, go to the original creator.