christianhaitian/arkos #346: Ir-MUPEN64PLUS (Retroarch 64 bits core) - won't boot

eckert2021, 2021-04-26:
Ever since the **04032021 v1.6** update, which supposedly  "_replaced retroarch 64 bit glupen64 core with Mupen64plus 64 bit core_" I cannot find a way to start any game with the MUPEN64PLUS core selected. It behaves just as it used to when the glupen64 core was "available", like it is simply a non existent core. 

I have even tried to load the core directly from the Retroarch 64 app directly, but it just won't load as well. I waited then for a few more FW updates to check if this bug would be corrected or if anyone else would report it. But since I'm still having this issue, I think now is a good time to report it, especially when no more updates for the RG351P are to be released.  

Thanks in advance!

christianhaitian, 2021-04-26:
I can load Blast Corps and Mario Kart 64 fine on my end with this core.  I just make sure they are in .z64 or .n64 format.  Someone else reported this issue in a Discord forum today but I can't replicate the issue as it starts just fine for my roms.  I confirmed this to still be the case by loading a fresh image, doing an OTA update to the latest version, and my roms still work with this core.  My guess is that the core fails to load due to some kind of incompatibility between that core and your roms.  As the core is deprecated and hasn't been updated since 2019(https://github.com/libretro/mupen64plus-libretro), I wouldn't be surprised about an incompatibility issue with some types of roms.

closed 2021-04-26 by christianhaitian

eckert2021, 2021-04-27:
Well, the thing is all my roms are either .z64 or .n64 and I tried with all of them. 
But that's a interesting idea: I'll try to write a fresh image in a brand new sd card to check if this awkward  issue still maintains.

eckert2021, 2021-05-16:
So I flashed a brand new image to my sdcard and it actually did work. As it turns out, you CANNOT update this core, otherwise it won't boot any game after that (I did several tests to come to this conclusion). My advise is to "lock" this core from any update or modifications over the respective retroarch settings.
If anybody is having the same issue, here is a link to the ".so" file of the Mupen64Plus core that actually works (which already comes with the final version of the ArkOs firmware if you did not update your cores). Just replace it in the appropriate Core directory:

[https://drive.google.com/file/d/1_Nm5uY62p41WlzIeCftMil_arbTTz3ba/view?usp=sharing](url)

christianhaitian, 2021-05-16:
I've confirmed this issue.  Since the original source for this core hasn't been updated since 2019, I went ahead and replaced the non working core in the repo with the one that works from this image.  In case someone else updates it, it will be one that works.  Thank you for mentioning this.