christianhaitian/arkos #347: RG351V Volume Buttons Not Working

joepistachio, 2021-04-27:
The volume button does not function on the RG351V build. Not sure if anyone else has seen this issue.

christianhaitian, 2021-04-27:
First report I’ve seen on this.  Works fine on multiple images I’ve tried so far.  

1. Have you updated online to the latest version?  If not, please do and test again. 
1. Do your volume buttons work on the stock firmware?  If not, your unit may be faulty.
1. Did you make sure to not use Balena Etcher to write the image?  If not, reimage with USB IT or win32diskimage.
1. Are you using a good quality SD card such as an authentic Samsung or Sandisk card? If not, please use a good quality SD card.
2. Have you tried to reimage your card yet? If not, try a fresh image.

christianhaitian, 2021-04-27:
Have you updated online from the options section?  You're ArkOS distro version should be 1.8.2 when you press start.

joepistachio, 2021-04-27:
I am currently using the Test Candidate OaD 1, is there a link to download an updated version of the rg351v firmware?  I haven't been able to find a different link anywhere for this.  I added a picture of what I see on my device.

[image]

christianhaitian, 2021-04-27:
You just update it through wifi.  Jist make sure your unit is connected to a wireless signal and Go to the options section and go to update and follow the directions.

joepistachio, 2021-04-27:
The update function on the Test Release 1 don't seem to work.  Tried reflashing again and still no luck.  I can confirm that I am connected to the internet though, I was able to update cores, cheats, etc in retroarch.

https://youtu.be/BYVYnHpr-FM (video of what happens when trying to update)

christianhaitian, 2021-04-27:
Very odd.  First i've seen that issue.  Try enabling remote services before attempting the update and see if that makes a difference.    If not, wait until the final image is available so you can try that one as there's been many updates since Test Release 1 was provided.  You can view the changelog [here](https://github.com/christianhaitian/arkos/blob/main/changelogs/rg351v-changelog).

joepistachio, 2021-04-27:
I tried to reimage it again this afternoon and for some reason it started working properly.  Thank you for all of your help, as well as all that you do to make these devices great!

closed 2021-04-27 by joepistachio