christianhaitian/arkos #350: Korean support is needed.

k-seijoo, 2021-04-28:
Korean language support is required for the RG351P.  In the meantime, there was no Korean font, so I used a Korean font.  However, since the last update, the Korean language is not displayed and is displayed as a square.  Official Korean language support is required.

christianhaitian, 2021-04-28:
Try to make sure [this folder](https://github.com/christianhaitian/EmulationStation-fcamod/tree/master/resources/locale/ko) exists in /usr/bin/emulationstation/resources/locale folder.  Otherwise, I don’t know how to resolve this.

closed 2021-04-28 by christianhaitian

k-seijoo, 2021-04-30:
Thanks for the reply.  And sorry for the late answer.  There are files in the folder you mentioned.  However, Korean is not displayed normally.