christianhaitian/arkos #359: Tweak System Configuration and Run Command sections

farfenkugell, 2021-05-01:
**Settings section:**
Not sure if possible, but it would be nice, to have "Enable remote Services" and "Update" to be set as Favorites out of the box after install. 
I am pretty sure these two are the most uses things, and for me, they are shown at the top.

"disable low battery warning" - it is not in caps, unlike every other option, maybe change it to "Disable Low Battery Warning"

Shorten some of the options:
"Restore Default Retroarch Settings" - "Reset Retroarch(64) Settings"
"Restore Default Retroarch Core Settings" - "Reset Retroarch(64) Core Settings"
"Restore Default Retroarch32 Settings" - "Reset Retroarch(32) Settings"
"Restore Default Retroarch32 Core Settings" - "Reset Retroarch(32) Core Settings"
"Restore Settings" - "Restore ArkOS Settings"

**Run Command section:**
It would be nice to rename the whole section to just Libretro at etc/emulationstation/es_systems.cfg

It also could be nice to add some minor description and stock art for retroarch 32/64 too:
https://imgur.com/a/AAOqVQW

`	<desc>RetroArch is a free, open-source and cross-platform front-end for emulators, game engines, video games, media players and other applications.</desc>
		<image>/roms/launchimages/ra32.png</image>`

`	<desc>RetroArch is a free, open-source and cross-platform front-end for emulators, game engines, video games, media players and other applications.</desc>
		<image>/roms/launchimages/ra.png</image>`

christianhaitian, 2021-05-01:
Most people don't know what Libretro is.  Better chance they've heard of Retroarch.

christianhaitian, 2021-05-01:
Not going to set  "Enable remote Services" and "Update" as that could potentially cause more support questions like the low battery indicator does for the rgb10, rk2020, and OGAs.  I'll let the individual users decide on that for their use case.  As an example, even though I regularly use those 2 features, I expect to only see games I regularly play in Favorites.

christianhaitian, 2021-05-01:
I've accommodated the following with today's update

- [ ] Renamed "disable low battery warning.sh" "Disable Low Battery Warning.sh"
- [ ] Renamed "Restore Default Retroarch Settings" - "Reset Retroarch Settings"
- [ ] Renamed "Restore Default Retroarch Core Settings" - "Reset Retroarch Core Settings"
- [ ] Renamed "Restore Default Retroarch32 Settings" - "Reset Retroarch32 Settings"
- [ ] Renamed "Restore Default Retroarch32 Core Settings" - "Reset Retroarch32 Core Settings"
- [ ] Renamed "Restore Settings" - "Restore ArkOS Settings"

closed 2021-05-01 by christianhaitian