christianhaitian/arkos #360: Old version

ashencos, 2021-05-01:
where I find iso of 1.5 was much more stable games were better.

christianhaitian, 2021-05-01:
Which system and games are you seeing issues with.  Please provide specific examples.

ashencos, 2021-05-01:
on arcade and etc already the loads are slower, and the games are not smooth but a little slow.

christianhaitian, 2021-05-01:
Have you tried a fresh image of 1.6 that was posted this morning to be sure it's not your current install?

ashencos, 2021-05-01:
I have installed the final

christianhaitian, 2021-05-01:
Well this is the first I'm hearing about poor performance since 1.6 came out.  As long as you're not using fast forward or rewind features in retroarch my guess would be a configuration tweak needs to be done to improve the gameplay for you.  You can try going into retroarch and Settings>Video>Threaded Video>On and Settings>Video>Synchronization>Sync to Exact Content Framerate>On and Settings>Video>Synchronization>Vertrical Sync (VSync)>Off and see if that improves the gameplay performance for you.  

If that doesn't work for you, you can try restoring the older version of Retroarch from [here](https://github.com/christianhaitian/arkos/issues/293#issuecomment-799774960).  That would be the only other reason you're experiencing a performance issue with 1.6.  No other differences would cause such an issue.

Also be sure you did not use Balena Etcher to image ArkOS onto your sd card and that your sd card is a good authentic brand of either Samsung or Sandisk.  These are common issues that lead to poor and unstable performance on this chipset.  Don't use the no name or unknown sourced sd cards that typically come with these handhelds.  

Good luck.

closed 2021-05-01 by christianhaitian