christianhaitian/arkos #363: All emulators with black screen when starting - RG351P on FINAL VERSION

AlexKCN, 2021-05-03:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

All emulators with black screen when starting

AlexKCN, 2021-05-03:
[image]
After this screen

christianhaitian, 2021-05-03:
I flashed the final image onto a good authentic sd card, (Samsung EVO 128GB) using USB IT.  After successful completion of the initial boot setup process on my RG351P, I tried the following systems and games:

**CPS3**
Street Fighter 3: 2nd Impact (Loaded fine)
Street Fighter 3: 3rd Strike (Loaded Fine)

**GBA**
Drill Dozer (Loaded Fine)
Super Mario Advance (Loaded Fine)
Super Mario Advance 2 (Loaded Fine)

**Neo Geo**
Baseball Stars 2 (Loaded Fine)
Sengoku 2 (Loaded Fine)

**PC-Engine CD-Rom**
Double Dragon 2 - The Revenge (Loaded Fine)

**PSX**
Bloody Roar 2 (Loaded Fine)
Mr. Driller (Loaded Fine)

For Neo Geo, PSX, and PC-Engine CD-Rom, I made sure their respective bios files were in the roms(easyroms)/bios folder.

I tried this with both the stock please wait picture and with the ascii art wait screen.  Never a black screen on the my end.

- [ ] Make sure you've followed the directions on the main wiki page, **closely** on how to properly load Arkos onto the SD card.  Make sure you **do not using Balena Etcher** as it causes oddities with ArkOS. (https://github.com/christianhaitian/arkos/wiki#instructions-for-loading)
- [ ] Make sure you're using a **good quality SD card like an authentic Sandisk or Samsung SD card**.  **Do not try to reuse the stock micro SD** that comes with the unit as those tend to fail quickly as they are of unknown quality.
- [ ] Make sure your roms are reliable.  Try them on other systems to make sure they work.

closed 2021-05-03 by christianhaitian

SuperBadger77, 2021-05-03:
Hi - just for an FYI - the link you've posted doesn't work: https://github.com/krishenriksen/RockRiotMenu

christianhaitian, 2021-05-03:
Fixed the template so future submissions should be correct.  Thanks