christianhaitian/arkos #365: Update from TR 2.1 to 2.2 failed

Gordon1972, 2021-05-04:
Hi there,
tried to update my RG351V from Test Release 2.1 to 2.2.
This was a rather  large update covering many files to download.
The update failed, because a downloaded file was broken.

Don't know how much free space is needed on the TF-Card, but i only had 1 GB free.
Anyway the file is broken and i can't install the update.

Is there anything i can do to delete the update-files on the card or something to retrigger the update?
I don't wan't to start from scratch (Test Release 1) again...

Thanks for reading.

closed 2021-05-04 by Gordon1972

christianhaitian, 2021-05-04:
Did you close this by accident or figure something out?

Gordon1972, 2021-05-04:
I figured out that you mentioned in another issue to enable WiFi or a related type of thing.
It worked, since after the failed attemp to update to 2.2 my Wifi was not enabled anymore.
So i chose to close the issue, since it really isn’t an „issue“ at all.

Anyway, thanks for getting back to me.

Greets

Von: christianhaitian
Gesendet: Dienstag, 4. Mai 2021 15:59
An: christianhaitian/arkos
Cc: Gordon1972; State change
Betreff: Re: [christianhaitian/arkos] RG351V: Update from TR 2.1 to 2.2 failed(#365)

Did you close this by accident or figure something out?