christianhaitian/arkos #366: do you play slow arcade games?

ashencos, 2021-05-05:
but is it normal for arcade games to be slow with everything running at 60 fps? that is, I remembered them faster, am I wrong?

ashencos, 2021-05-05:
why it often slows me down even audio slows down etc can 'depend on the roms or the defective console?

christianhaitian, 2021-05-05:
I have no slow downs on the arcade games I play.  I use those that are compatible with fbneo for best performance. MAME ones can have slow downs sometimes depending on the romset and other factors. I recommend that you join the RGHandheld discord and get help there where there's a large community to assist with your questions about ArkOS.  Link is https://discord.gg/wurh4WM

closed 2021-05-05 by christianhaitian