christianhaitian/arkos #370: Changing cores per game or per system

huguesmp, 2021-05-08:
* Unit: RG351V
* Os Version: Test Rlease 2.4 (Installed as 2.1 and updated using the update fonctionnality in options)

After getting some slowdown in Yoshi's island, I decided to switch the core to SNES9x 2010. Even though the core is present (it has a # symbol in the RetroArch Core Installer), I don't see it as a choice in EmulationStation, either in emulator settings or in the game's "edit metadata" option.

Apologies if I'm missing something obvious, and a big thanks for this amazing project.

[image]
[image]

christianhaitian, 2021-05-08:
https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V#a-do-the-following
You must first move the emulator from default to retroarch to be able to select a different core.

closed 2021-05-08 by christianhaitian

huguesmp, 2021-05-08:
Worked like a charm, thanks!