christianhaitian/arkos #371: Simple rom delete

Sanesecurity, 2021-05-09:
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

Feature addition

Could a simple delete rom option be added to the gui. 

When you browse the List of games in the normal way and see one you don't want or it's duplicated.. Have an option to delete it from the list... without having to
go to file manager to try and work out what filename it is. 

### How can the issue be reproduced? (If applicable)

### What device are you using?

Rg351v

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

Latest 

### Anything else you'd like to include that may help to review this issue or feature request?

Thanks for the rg351v support 👏👏👏

christianhaitian, 2021-05-09:
This is already possible.  While on the rom you no longer wish to have on the sd card, hit select, then edit metadata, and select the delete option at the  bottom of the menu next to cancel.

closed 2021-05-09 by christianhaitian

christianhaitian, 2024-05-30:
Do that before you load the roms onto the sd card.  It would be much faster and efficient to do that prior to loading them on to the sd card.