christianhaitian/arkos #373: Feature request: m3u filter toggle for psx

farfenkugell, 2021-05-15:
Consider adding "Script" to adjust what formats are displayed in some of the systems.
For example, pretty much the only system that is impacted for me is PS1 games.

I use .chd that are working fine, the problem arises when the game has multiple disks.
Fix is simple, use .m3u, but the problem is that you now have entries for each disk+m3u listed after that.

In some other OS's you can set a filter to define which files to display, thus I have m3u created for each game, even single disc games.

In ArkOS, to my knowledge, you need to edit:
etc/emulationstation/es_systems.cfg

`<name>psx</name>` 
`<fullname>Sony - PlayStation</fullname>`
`<path>/roms/psx/</path>`
`<extension>.m3u .M3U</extension>`

So pretty much an entry in Options - "PS1 - Show only m3u games" / "PS1 - Show all games".
That would modify the "extensions" part of the config.

*Bonus points would be to have a script that creates these m3u files from ArkOS itself.
**Multiple disc games in my environment are in sub-folder, single disk games are in the root folder of psx

christianhaitian, 2021-05-15:
This would require extensive modification to emulationstation that I'm not willing to invest time in for this.  You're welcome to review the emulationstation source used in ArkOS [here](https://github.com/christianhaitian/EmulationStation-fcamod) and submit a PR for the functionality.

christianhaitian, 2021-05-15:
I now realize that the request is just to add the ability to change this option from a script in the options menu.

farfenkugell, 2021-05-15:
`for file in *.chd; do echo "$file" >> "$(basename "$file").m3u"; done` on linux

`@echo off
for /R "C:\Users\vaido\Downloads\Test\" %%I in (*.chd) do for %%J in ("%%~dpI.") do echo %%~nxI>>"%%~dpI%%~nxJ.m3u" ` on windows via .bat

christianhaitian, 2021-05-18:
This is a handy feature and I have found a script (https://github.com/danyboy666/Generate-PSX-m3u-playlist/blob/master/generate_psx_m3u.sh ) that I’ve modified to handle .m3u creation even for subfolders.  Thanks for helping to test it.

christianhaitian, 2021-05-20:
This has been accommodated with today's update for the RGB10, RK2020, and the OGAs.  Will also be available in the next update for the RG351V.

closed 2021-05-20 by christianhaitian