christianhaitian/arkos #374: Feature request: More Options Menu Cleanup

farfenkugell, 2021-05-15:
While naming is understandable in pretty much all options now, making slight adjustments it would allow to make it even better.

"Enable Remote Services" and "Disable Remote Service", both are clear, but they are not going one after another.
How about renaming them to "Remote Services - Enable" and "Remote Services - Disable", this way they would go in a row and make things simpler.
"Backup ArkOS Settings" and "Restore ArkOS Settings" to "ArkOS Settings - Backup" and "ArkOS Settings - Restore", same words, same meaning, but again, they would go in a row and make things look cleaner.

"Disable Low Battery Warning" and "enable low battery warning" to "Low Battery Warning - Enable" and "Low Battery Warning - Disable".

"ES FullScreen" and "ES Header" do go in a row, but as there are other options that take more symbols, I see no reason why they could not be renamed to "EmulationStation FullScreen" and "EmulationStation Header" for clarity. 

Regarding all of the stuff related to RetroArch entries (in options), not sure how popular those are, but if possible (simple?), how about merging all of those entries into a single one that would open terminal like interface in which you can select which option to use (similar to the one that RetroPie has for optional package management) and call it just "RetroArch Reset" or maybe even moving all of them to RetroArch/Run Command if it is possible to make it that RetroArch 32/64 still appear as first two entries in that list.

christianhaitian, 2021-05-18:
My concern for this is that if I change this up, then others will say well now they have to scroll farther down for options they may use regularly.  I’ll leave this ticket here for now and see if anyone else would like to see these changes or to just leave them alone.  Maybe most don’t care and if not, I’ll make these changes.  Not really wanting to spend to much time on these items so I don’t want any back and forth on something like this if I can help it.

farfenkugell, 2021-05-18:
Using "Favorites" allows having entry listed at the very top. I personally have "Update" and "Enable Remote Services" in favorites, thus they always appear at the top and they are the only options that I use.
On top of that "System/Collection" Favorites is hidden in my ArkOS installation and I use "Last Played", this all depends on how it is configured in other people's OS.

Personally, the suggestions above would make things clearer and more simple to navigate.

taivu, 2021-05-20:
fwiw, throwing in my 2 cents: i really like @farfenkugell 's suggestions personally, but i think in some themes it would be hard to see. some themes cut off the text in the game list, so youll end up seeing "Low Battery Wa" twice.

molcos77, 2022-07-21:
Or do something similar for most CLI based linux/Unix systems that have you select items from a list.

Number them...

1. Whatever xxx
2. Whatever again xxx
3. Still here???

That way you have complete control over what goes where as it's sorted by numbers instead of letters.

christianhaitian, 2023-01-21:
I've decided that this is not something I will focus on anytime soon.

closed 2023-01-21 by christianhaitian