christianhaitian/arkos #375: Corrupted global hotkeys (event keys) on RGB10 and restore method

fakemaria, 2021-05-15:
### Describe the Issue (If applicable)

I have found that for some reason, the combination of + button and up, down, left and right didn´t work properly and I can´t change brightness and volume. This is a problem since the process of backup all and restore all is painful and I would like to know which file do I need to replace or which line governs this controls. 

### How can the issue be reproduced? (If applicable)

As described before!

### What device are you using?

RGB10

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

1.6 

### Anything else you'd like to include that may help to review this issue or feature request?

Thanks in advance for your hard work and sorry to raise a bug, this request is more troubleshooting rather than a bug and maybe the system got corrupted somehow. But is weird that the rest of the functions with minus (-) buttons are working as expected.

christianhaitian, 2021-05-15:
A reboot should return the function of the global hotkey buttons.  I plan to provide a solution in the options/advanced to fix that in rare occasions when it occurs and a reboot doesn't fix it.  If you're comfortable with ssh, you can also ssh into the unit and do 

`sudo systemctl enable oga_events && sudo systemctl restart oga_events`

closed 2021-05-15 by christianhaitian

fakemaria, 2021-05-16:
Its weird but it didn´t fix the issue and can´t use the + button to modify brightness and volume :/

christianhaitian, 2021-05-16:
How long have you had this issue?  Did that ever work for you?

fakemaria, 2021-05-17:
since I installed the system. Never worked for me, the combos with left (-) minus buttons works ok. 
I was thinking that maybe it was the button wasn´t properly setup or was faulty, but I´ve launched an evtest and seems to be recognized. 

I´ve updated the system to 1.7 and same result.

christianhaitian, 2021-05-17:
Are you sure you loaded the right image?