christianhaitian/arkos #376: Pico-8 logo not showing in es-theme-arkos-carbon

SerjTargarien, 2021-05-16:
There is a problem with the Pico-8 logo in the ArkOS Carbon theme.

pico8.png should be pico-8.png in ./roms/themes/es-theme-arkos-carbon/art/controllers. 

The default naming is wrong so it doesnt show up properly in EmulationStation.

taivu, 2021-05-20:
@SerjTargarien looks like the theme's repo is updated with that file name change: https://github.com/Jetup13/es-theme-arkos-carbon/commits/master

christianhaitian, 2021-05-21:
This has been resolved and you can now use ThemeMaster by JohnIrvine to update your themes and install new ones.  You can even see previews of the themes as well.  Check it out at https://github.com/JohnIrvine1433/ThemeMaster.  It’s compatibles with all devices that ArkOS works on as of today.

closed 2021-05-21 by christianhaitian