christianhaitian/arkos #382: Swap A/B buttons in EmulationStation

Pacoboyd, 2021-05-21:
Looking for an option to switch the A/B buttons in EmulationStation much like retroarch has for their menu navigation. I'm an SNES guy and having A for my confirm really messes me up.  In the non arkos version within developer options there is a way to swap it normally but I'm not able to find a way to do so with ArkOS.

christianhaitian, 2021-05-21:
I disabled that in the Emulationstation for ArkOS as reconfiguring buttons in the menu for Emulationstation used for ArkOS, removes the page up and page down feature.  If you are handy with ssh or winscp, you can go to /etc/emulationstation/es_input.cfg and swap input name="a" and input name="b" which should give you what you want without removing the page up and page down features.

closed 2021-05-21 by christianhaitian

Pacoboyd, 2021-05-21:
Thanks, the work is done.  Might be worth throwing in the FAQ as I did actually try and search (here and reddit) before I posted an issue.  Thanks for the quick response!