christianhaitian/arkos #384: Scraper (bug)

eckert2021, 2021-05-22:
I am not being able to scrape any of my games from the “screenscraper.fr” plataform, only by “thegamesDB”, which sadly does not scrape the videos. 
Funny thing is that apparently the names of the roms are scraped, but the images and videos are not downloaded. I am also having this issue with my 351ELEC sd card.

 I dont know if this bug is affecting only the 351P/M devices or any device trying to scrape from “screenscraper”.

Does anyone have a clue? Thanks in advance.

ps: I even tried to login with a different account.

michaeltzinn, 2021-05-22:
I had the hardest time until I actually registered on the screenscraper.fr site. I thought I was registered but after many frustrating attempts I finally saw that well below the register dialogue box there was some info that was required to complete my registration.  Now only takes a few restarts to scrape. However I probably had it 90% complete by then. Maybe check registration.  Hope you get it resolved.

christianhaitian, 2021-05-22:
It seems to be working again on my end.  Although, I can't see previews of the images like you could previously when you try to scrape just 1 game via the Edit Metadata menu.

christianhaitian, 2021-05-22:
Now the previews seem to work in the Edit Metadata menu.  Maybe server load or other congestion issues.  Closing this ticket as the issue is not an issue within ArkOS.

closed 2021-05-22 by christianhaitian

eckert2021, 2021-05-22:
> I had the hardest time until I actually registered on the screenscraper.fr site. I thought I was registered but after many frustrating attempts I finally saw that well below the register dialogue [...]

Trying and retrying to scrape each game individually was actually the way to go. The attempt is going to be successful as soon as the preview image decides to show off.
Thanks for the tip!