christianhaitian/arkos #385: 640x480 resolution for Atomiswave and Naomi like Dreamcast

beatlesdbp, 2021-05-22:
Would you be able to do this for Atomiswave and Naomi as well? I set the Atomiswave core as retrorun32 and held down the A button while loading but the game did not play in 640x480 resolution.

TRv1.6 (04/18/2021)
-Finally fixed issue with enabling 640x480 with retrorun32 for dreamcast. 
  (Just hold A while launching a game to enable 640x480.  You should see "640x480 enabled" on the screen while launching.
-Added video player.
  See the ArkOS wiki here (https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V#q-how-does-the-movievideo-player-work) for controls and other info.

christianhaitian, 2021-05-22:
Sure.  I’ll add this in the next update.

beatlesdbp, 2021-05-23:
Awesome! Thanks so much!

christianhaitian, 2021-06-05:
This has been accommodated with today's update.

closed 2021-06-05 by christianhaitian

beatlesdbp, 2021-06-05:
You are the man! Thanks so much!!!