christianhaitian/arkos #387: .BS games do not open via the emulationstation snes-hacks menu but they boot if you do it manually from the retroarch menu.

megamanaso, 2021-05-24:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/AnberPorts) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-05-24:
What emulator did you use when you did it manually in retroarch?

megamanaso, 2021-05-24:
snes9x2010.  I also set that as the emulator for snes from the emustation but it wouldn't boot that way either.

duskgao, 2021-05-24:
Satellaview games need you put the bs-x.bin in your bios dir.

You can find it at https://project.satellaview.org/downloads.htm

Remember,you should use the no drm version bios

megamanaso, 2021-05-24:
Yep thats the bios i had already there. I replaced it just in case and im having the same issue. I tried all cores from the emustation menu but it still doesnt boot. It flashes the ARKOS screen and it goes back to the game list. I tried it directly from retroarch and it still works just fine that way.

christianhaitian, 2021-05-24:
Hmmm....let's see.  If it works from Retroarch, it should work from ES as well.

christianhaitian, 2021-05-24:
I found the issue.  .bs files can only be loaded with snes9x as @duskgao suggested as a resolve above.  They don't seem to work with snes9x2010.  I will include it as an available emulator for snes-hacks in the next update.

megamanaso, 2021-05-24:
Cool thank you for talking a look at it. Do you have a time frame for the next update by any chance?

closed 2021-05-24 by megamanaso

christianhaitian, 2021-05-24:
probably early June.

megamanaso, 2021-05-24:
Cool sounds good and 1 last thing I haven't been able to find. Where is update game list button ?

christianhaitian, 2021-05-24:
Update game list button?  No such thing.  Hit start, then go to quit then restart emulationstation.  That will refresh the game list that way if you haven't told ES to just parse the existing gamelist only.

megamanaso, 2021-05-25:
Yep thats what i have been doing. I deleted the drastic config and backup today by mistake. Is there anywhere specific i can find that on your github?

christianhaitian, 2021-05-26:
You can use 7zip and open the image and grab it from there.