christianhaitian/arkos #388: EasyRPG runs way too fast (RG351V)

MaxiMuM1441, 2021-05-25:
Hello! As the title says RPG Maker games run a lot faster than they should. Im running the latest version of ArkOS on my RG351V (EasyRPG v0.6.1). I've tested multiple games and they all share the same issue. This problem is already visible when I start the EasyRPG Player itself and not load any game. The selection box on the left side blinks like at least 4x as fast as it should. I hope I'm not the only person with this issue and this can be fixed in a next update. Thanks!

christianhaitian, 2021-05-26:
Hello,

Please follow the instructions below to test this updated version of EasyRPG and provide feedback on this when you can to see if this resolves the issue.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device and test EasyRPG.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6543590/arkosbackup.tar.gz)

MaxiMuM1441, 2021-05-26:
This newer version of EasyRPG fixes this issue. Games run at the correct speed now but it comes with new problems. It says that the WAV format is not supported so there is no sound at all. And you can't reset the game with the select button. I will do some more testing to see what else is not working.

MaxiMuM1441, 2021-05-26:
It plays midi files fine but they sound really weird. It's not using the same instruments as in the previous version to play those files. I don't know if that is something that you can fix but they sound pretty bad.

christianhaitian, 2021-05-26:
Following the same steps from before, restore this one and see if this is better.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6543901/arkosbackup.tar.gz)

MaxiMuM1441, 2021-05-26:
Now it plays WAV and MP3 files without issues. The select button is still not resetting the game. A window pops up that says "Restarting in 2 sec ..." for a very short amount of time but thats all.

christianhaitian, 2021-05-26:
Apparently select is not supposed to reset the game.  It was bug.  Select + B is supposed to reset the game which seems to work on my end.

christianhaitian, 2021-05-26:
I also noticed if you move the reset function to another key like L or R, holding the button down for 2 seconds, restarts the game.

MaxiMuM1441, 2021-05-26:
> Apparently select is not supposed to reset the game. It was a bug. Select + B is supposed to reset the game which seems to work on my end.

It is better like this anyways. So it doesn't reset the game when you accidentally hit Select.

Thanks for your work man!

MaxiMuM1441, 2021-05-26:
About the midi files. It's still using the wrong set of intruments to play those. Is it possible for you to change them back to the original ones?

christianhaitian, 2021-05-26:
I am not sure this is something I can fix but I can review it.  The one game I tried named Lufia-Idura's Revenge 1 seems to sound fine to me.  Is there another game I can try and compare with the previous version of easyrpg and this new one so I can better hear the difference?

MaxiMuM1441, 2021-05-26:
I tried multiple games and they share the same problem with mifi files sounding not like they should.
Here is what I mean. This is how it sounds like: 
https://soundcloud.com/maximum01/bad/s-Hy4TUlmLwE6

This is how it should sound like (EasyRPG on Windows):
https://soundcloud.com/maximum01/good/s-HiRb8ODOMVf

I'm pretty sure this is how it sounded like under v0.6.1 under ArkOS but at this point I'm not quite sure anymore.

The game I tested is Vampires Dawn 2 - Everlasting Blood
https://www.vampiresdawn.org/forum/viewtopic.php?t=40624

christianhaitian, 2021-05-26:
Both of these don't sound like midi at all.  They both sound like different qualities of synthesizing.  Try restoring this file, this is the old 0.6.1 EasyRPG version that was in ArkOS and see if sounds better.  Let me know and I can try to investigate if there's a truly a difference.  

Thanks,
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6550042/arkosbackup.tar.gz)

MaxiMuM1441, 2021-05-26:
Ah my bad. Both version sound the same. I wish it it would sound like EasyRPG on Windows. I guess this is something that you can't fix then.

christianhaitian, 2021-06-05:
This has been pushed in today's update.

closed 2021-06-05 by christianhaitian