christianhaitian/arkos #389: a manual way to install an update?

mariotaotao, 2021-05-26:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/AnberPorts) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Hi, thank you so much for your contribution on this handheld device.
I always can't do online update successfully, I have tried many ways, using hotspot with my phone, changing different ISP networks, trying to edit /etc/environment or /etc/profile file to add an http profile (but I've failed to mod them)
some times it give me different error, I've already try this https://github.com/christianhaitian/arkos/issues/311 method to work around, but that did not work. 
### How can the issue be reproduced? (If applicable)
anytime when I try to update

### What device are you using?
RG351V

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
1.7 without any update

### Anything else you'd like to include that may help to review this issue or feature request?
hoping to have a manual way to install the updates, I think it will be very convenient if you can add a proxy menu on wifi feature.

christianhaitian, 2021-05-26:
Not at this time.  If you can't update, it could have been an issue with github as well.  May have to try at a different time.

closed 2021-05-26 by christianhaitian

mariotaotao, 2021-05-30:
@christianhaitian
Ok, thanks.
I have another question, how to update the cores of retroarch?

christianhaitian, 2021-05-30:
retroarch menu>online updater>update installed cores

mariotaotao, 2021-06-09:
Hi @christianhaitian , do you have plan to release a new image that contains all the latest updates? Comparing to other works I prefer to use Arkos, but can't do an updates successfully. Thanks

christianhaitian, 2021-06-09:
For the 351V, the last image was created a few weeks ago so it will be awhile yet.

mariotaotao, 2022-02-26:
Hi @christianhaitian, I notice the latest update can't be done by switching to China server. Can you check it? Thanks

mariotaotao, 2022-02-27:
@christianhaitian
I have tried add proxy in /etc/environment  
http_proxy="host:port"
https_proxy="host:port"

but update still direct to China server, the proxy method worked before.

christianhaitian, 2022-02-27:
Yea the update script goes by your geolocation.  If you’re located outside of mainland china, you can modify the `/opt/system/Update.sh` and change the `clocation` variable value to what’s shown for `location` and that should work to update.