christianhaitian/arkos #392: Request: make wifi toggle on/off persistent throughout reboots

wouterVE, 2021-06-07:
### Situation now
When you toggle off wifi & reboot your device, wifi is toggled on again.
So basically you have to manually turn off wifi every time you boot the device.

### Desired situation
Make wifi toggle on/off persistent througout reboots so when you toggle wifi off, it remains off until you manually decide to turn it on again.   This will also save battery power. 

### What device are you using?
Odroid Go Advance 1.1

### What version of ArkOS are you on?
v1.7 (06/04/2021)

christianhaitian, 2021-06-07:
How are you toggling wifi off on the OGA 1.1 with ArkOS?  I never added that feature for the OGA, rk2020, or rgb10.

christianhaitian, 2021-06-07:
Maybe through the wifi script in the Option section?  I'll see about adding an ability for wifi to be toggled specifically for the OGA 1.1.

christianhaitian, 2021-06-07:
Hello,

Please follow the instructions below to test the Wifi-Disable and Wifi-Enable scripts and provide feedback on this when you can to verify the function as expected.  **This will only work on ArkOS loaded on a OGA 1.1 device only!**:

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, restart Emulationstation or restart your device and you should now have 2 options available in the options/Advanced section.  One named Wifi-Disable and another named Wifi-Enable.  Launch Wifi-Disable and verify that Wifi is indeed turned off and that it is still off on reboot.  Then launch Wifi-Enable and verify that Wifi is indeed turned on and that is is still on on reboot as well.

Again, feedback on this would be appreciated.

Thanks

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6612207/arkosbackup.tar.gz)

wouterVE, 2021-06-08:
> How are you toggling wifi off on the OGA 1.1 with ArkOS? I never added that feature for the OGA, rk2020, or rgb10.

Hello, 
See attached picture: 
[image]
I toggled this off using the WIFI application & then selecting my SSID and pressing deactivate.

I've tried your scripts and can confirm that it works!

- Wifi-Disable: wifi is turned off immediately & persists through reboot (using network-info  returns to emulation-station immediately)
- Wifi-Enable: wifi is turned on immediately & persists through reboot

Thanks for your quick action! Just wondering, I've taken a look at your scripts & how did you achieve this? By de-activing the kernel modules for the wifi-chip in the odroid I suppose?

kr

Wouter

christianhaitian, 2021-06-08:
Excellent.  And that's correct about how wifi disablement persists.  Below is what's contained in each script.  The nmcli statement is commented out but maintained in the script in case I want to use that again the future.

Wifi-Disable.sh
```
#!/bin/bash
sudo modprobe -r esp8089
sudo sed -i '$ablacklist esp8089' /etc/modprobe.d/blacklist.conf
# sudo nmcli r wifi off
printf "\n\n\n\e[32mWifi has been disabled.\n"
sleep 1
```
Wifi-Enable.sh
```
#!/bin/bash
sudo modprobe -i esp8089
sudo sed -i '/blacklist esp8089/d' /etc/modprobe.d/blacklist.conf
#sudo nmcli r wifi on
printf "\n\n\n\e[32mWifi has been enabled.\n"
sleep 1
```

wouterVE, 2021-06-09:
Yes, I was also thinking about a more general solution for other devices (I suppose only OGA has built in WiFi atm).
For your reference:
`rfkill block wifi`
Should have similar results, just in case  `nmcli` is not working properly :-)

[source](https://manpages.ubuntu.com/manpages/precise/man8/rfkill.8.html)