christianhaitian/arkos #396: Feature Request: Use both SD cards for storing games in 351V

sebvalmontx360, 2021-06-14:
### Describe the Issue (If applicable)
It is not a bug. It is, in fact, a feature request. Would it be possible to use both SD cards on 351V for storing roms? For instance, I would like to use two 128GB SD cards. The first one would store ArkOS itself and some systems; the second one would store other systems. In this case I would have almost 256GB for games.

### How can the issue be reproduced? (If applicable)
It is not a bug.

### What device are you using?
RG351V

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
1.7 for 351V

christianhaitian, 2021-06-14:
It's possible but you'd have to manually manage that.  Emulationstation will only read one path for games per system.  So you'd have to manually edit es_systems.cfg in /etc/emulationstation to change the location of where to find games.  It be nice to be able to update those locations from within emulationstation itself without manually editing the configuration file but it's not a feature I will be adding to emulationstation in the near future if at all.

closed 2021-06-14 by christianhaitian

sebvalmontx360, 2021-06-14:
Thank you for your answer. I totally understand that it is not a priority at this time. But you really gave me a good solution. Since I want to use one SD for some systems and other SD card for other systems, I can point each system to the correct SD card. It is a very simple task to do. Would my cfg file be overwritten in an update?

christianhaitian, 2021-06-14:
I try to do sed based changes to es_systems.cfg when possible.  I also usually create a backup copy of the existing es_systems.cfg file before making changes as well.  As good practice, you should create a backup copy of your own just in case any future updates overwrite a modification of yours or the OS or sd card corrupts.