christianhaitian/arkos #398: No Gamepad Detected

kiki0931, 2021-06-30:
Message "No gamepad detected" while booting RG351V 

Appears essentialy when battery level is low (but not only)

On last build 06042021 made online

Software or battery problem ?
Even by plugging in the device the message appears anyway

christianhaitian, 2021-06-30:
That’s a hardware issue.  This issue plagued the 351p/m units as well.  It can be caused by a low or bad battery, loose connection for the gamepad, or a faulty unit.  Try making sure the unit is fully charged, if it is and still occurs, you can try to open the unit and make sure the cables are fully seated.  All else fail, you may need a replacement unit.

closed 2021-06-30 by christianhaitian

kiki0931, 2021-06-30:
thanks a lot