christianhaitian/arkos #399: WiFi does not work

wifiissues, 2021-07-03:
Hi there, I have a WiFi dongle on your list (TP Link WN725N) and a working OTG adapter (tested on another computer) but for some reason, when I try to enable WiFi, I see no options for me to select a WiFi network and I'm not sure what I can do to turn the dongle "on" so to speak. Can you please give me some pointers? Love the rest of the OS, so thank you!

christianhaitian, 2021-07-03:
What is your unit?

wifiissues, 2021-07-03:
Sorry! I'm using a RG351P. When I plug in the WiFi dongle with OTG adapter, nothing happens but when I use them independently I get a response from the hardware.

christianhaitian, 2021-07-03:
Did you follow the instructions from here?  https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-configure-my-wifi-connection-in-arkos

wifiissues, 2021-07-03:
Yes, I read through that first. So, I removed the ethernet connection, then tried adding a new connection by selecting "+" but nothing shows up. It will let me do nothing but to exit by selecting "x".

christianhaitian, 2021-07-03:
Did you make sure to push the otg adapter all the way into the unit until it clicked?  Also, did you make sure to not have the unit charging as you attempt to connect your wifi adapter?  Make sure the wifi adapter is the only thing you have plugged into any of the usb c ports when trying this.

wifiissues, 2021-07-03:
Yessir, I actually started off with a faulty one and had to double check with other computers that the OTG adapter was indeed malfunctioning. Took it back to the shop and they confirmed and replaced my unit and I also confirmed at home that the unit was responsive by plugging it into a thumb drive.

I have been conscious not to be charging the RG351P at the same time as well. It seems my issue is not that common - which is slightly worrying. The good news is that it's working well but WiFi updates and scrapes for installed ROMs would be very welcome. I should have bloody bought the RG351M..!

christianhaitian, 2021-07-04:
You downloaded the final image from the github?  or are you using a preconfigured card?

wifiissues, 2021-07-04:
Hi there, yes - I'm running the final version of ArkOS. I started with a blank micro SD to install ArkOS

christianhaitian, 2021-07-04:
Ok.  So there are no further updates available besides retroarch core updates.   For wifi, not sure whatelse you can try unfortunately.

christianhaitian, 2021-07-04:
Does the adapter work with stock or 351Elec firmware?

wifiissues, 2021-07-04:
No worries and I appreciate your time.

It's just a really nice to have.

I don't have 351Elec but the stock card that came with the unit with Emuelec doesn't work either...

christianhaitian, 2021-07-04:
Sounds closer to a hardware issue

wifiissues, 2021-07-04:
Any suggestions? The WiFi dongle registers on a PC to install drivers when I plug it in without the OTG adapter.

The OTG adapter is able to read usb drives I mount with ArkOS.

Not sure what else I can try unless I can flash the dongle?

christianhaitian, 2021-07-04:
If it doesn't wok on stock or in ArkOS final, I have no other suggestions since I've dropped support for the P/M units.  Perhaps try 351Elec and see if it works there.  If not, perhaps that unit is not quite the same 725 unit as previous.  Maybe there was a revision to the chipset.

closed 2021-07-04 by christianhaitian

wifiissues, 2021-07-05:
Thank you for your suggestions. I guess I'll just have to keep an eye on whether anyone else has this issue later on since it's not essential and the emulators are working well.

Thanks so much for your willingness to discuss potential solutions!