christianhaitian/arkos #402: How to solve the horizontal linear display failure of 351V?

haveanair1, 2021-07-19:
Not all devices are like this, but a significant percentage of horizontal lines are occurring with the 351v. Pure Firm and ArcOS are both. We recently released a firmware that eliminates the streak in Anbernic. I would like to ask if there is a patch that can eliminate the failure with ArkOS. Here is a link to  Anbernic's patch firmware. https://drive.google.com/file/d/1ePdwJPUui6bUpXhv1crWfNu3dGfL5SA5/view

### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/AnberPorts) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-07-19:
As far as I know, those line issues on the screen are hardware issues.  This has been somewhat common on the 351V units and the only fix is to return them to Anbernic for repair or replacement.  I've had 2 units and none of my units had this issue with ArkOS. I still have one that I use for development and leave it on for many days at a time and never experience this issue.

YonKuma, 2021-07-19:
@haveanair1 I can't seem to find any mention of this link on any official sources. Where did you get this patch?

haveanair1, 2021-07-24:
The horizontal line phenomenon is a phenomenon that is often seen by those who buy a new 351V these days. In the beginning, this was not the case. It seems that the manufacturer produced by replacing parts in the middle and released it without completely solving the problem caused by it. The firmware shared above is the firmware I received by inquiring separately from Anvernic. On devices with horizontal banding, using the new firmware will improve the banding, but using the original firmware or ArcOS will cause the problem again.

YonKuma, 2021-07-26:
@haveanair1 Does it improve the banding, or does it completely solve it? Have you stress tested it by playing a PS1 game for a couple of hours using that firmware? It seems like heat/stress seem to make the issue worse.

This still seems likely to be a hardware issue, and unless the firmware is 100% effective in solving the problem, I think returning the device and trying to get a new one without the issue is the preferred solution.

christianhaitian, 2021-08-06:
Hello,

This DTB has the LCD fix from that source you provided #for ArkOS.  Just backup the existing one in the main sd card's boot partition, copy this one to the boot partition of the main sd card, and try it out and let me know if you experience any improvement or other differences during your testing.  

Thanks

christianhaitian, 2021-08-11:
I've gotten confirmation from multiple sources that this updated dtb seems to have addressed the black lines issue.  Closing this ticket and will provide it with a future update.

closed 2021-08-11 by christianhaitian