christianhaitian/arkos #404: RK2020 - There was an Error installing this update.

ZeiramPP, 2021-08-02:
Wegen trying to update from 1.6 to 1.7 i receive the error message named in the title. Remote services are enabled, time is set properly and WiFi dongle is working.

I was able to install the update before but the memory card died and i wanted to create a new one, but was not able to do this yet...

christianhaitian, 2021-08-03:
It may be github had issues earlier or maybe your internet connection.  I just loaded a fresh image of 1.6, updated to the 1.7 (07282021) without issue.  Only suggestion I have is to be sure your internet connection is stable, your not to far from your wireless signal and try it again.

christianhaitian, 2021-08-05:
Any update on this?

ZeiramPP, 2021-08-06:
Sorry, was busy due to my work. 🙂

The problem was my router somehow, because it decided to block out the RK2020, while other devices like the Odroid Go Super or the Piboy work without problems. After making some changes to the channels used by 2.4ghz WiFi it worked like a charm... 😊

christianhaitian, 2021-08-06:
Great news!  Enjoy!

closed 2021-08-06 by christianhaitian