christianhaitian/arkos #405: Gamefoce Chi Network Info not displaying

SuperBadger77, 2021-08-09:
2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ? Yes.

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ? Yes

### Describe the Issue (If applicable)
I have the latest test version of the Gameforce Chi image - Test Release 1.2. I am connected to the internet - can scrape. If I go to setup scripts and click network info, the screen changes, then reverts back to the scripts screen. No network info is displayed.

### How can the issue be reproduced? (If applicable)
Make sure wifi working. Setup Scripts -> Network Info: no network information displayed.

### What device are you using?
Gameforce Chi

### What version of ArkOS are you on?
Test Release 1.2

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-08-09:
I just checked this on my end and Network Info works.  Did you accidentally disable your wifi with the Home+Select key toggle?  From the start menu, you should see Wifi: up if you're connected to wifi.  If you do, Network Info should work.

christianhaitian, 2021-08-11:
Any update on this?

SuperBadger77, 2021-08-11:
Let me test it now.

SuperBadger77, 2021-08-11:
Yes - my bad - must have pressed home and select when trying to exit a game. Can close this one off.

closed 2021-08-11 by SuperBadger77