christianhaitian/arkos #407: Video preview not working

ciretatro, 2021-08-16:
I have the latest version of ArkOS running on a RG351P. Midusage my device turned off due to battery level. After charging the device, the OS would not boot up. I reformatted my SD card and reflashed ArkOS and did a clean install. Now video previews only work for 2 systems despite having all of the images and videos scraped for everything. Box art shows fine, it seems to be only for video preview. I transferred an identical copy of my SD card during the reinstall process and did not alter any settings. Is this a known issue (i could not find it there) and if not, could it be as simple as the proper file directory is not specified?

christianhaitian, 2021-08-16:
Sounds like the system can’t find the videos for your other systems.  Double check that your systems have a video folder for each of them within their respective rom folders.  If so,  You can either manual check the gameslist.xml file for each system that’s not functioning and verify that they’re pointing to the right video folders or rescrape the systems that are not playing videos.

ciretatro, 2021-08-16:
That's exactly it. It appears, somehow, all but few of my gamelist.xml files no longer have a video path specified. I now realize this wasn't a ArkOS issue but I appreciate the help on this, I would have been running in circles. I swear I spend more time in notepad ++ then using this damn thing!

closed 2021-08-17 by christianhaitian