christianhaitian/arkos #408: Request: Upgrade retroarch for RG351P/M

fishman8, 2021-08-20:
First I would like to say many thanks for your effort, arkOS is a grate firmware for RG351 unit.
Compared to EmuElec, 351Elec, batocera, I still like to use arkOS for good performance, ubuntu env, and apt-get !

Although I understand you already released the final version for RG351P/M, I still want to request the new version of retroarch. Build retroarch for the unit is too hard for me, please consider to update retroach for RG351P/M. 

Thank you.

Turro75, 2021-08-20:
You would try this fork which implements ra 1.9.6 and more, there are a couple of video on yt from wummle which explains the details
[https://github.com/wummle/arkos/releases](https://github.com/wummle/arkos/releases)

fishman8, 2021-08-20:
> You would try this fork which implements ra 1.9.6 and more, there are a couple of video on yt from wummle which explains the details
> https://github.com/wummle/arkos/releases

Thanks your information. After I checked the script and the update file, it's weird. He just updates .so files then makes retreoarch becomes to 1.9.6. If we check arkosupdate07022021.zip form here, it contains retroarch execute files. 

Hope christianhaitian can give the updates for RG351P/M. Thanks.

christianhaitian, 2021-08-20:
Hello,

Thank you for the compliment.  I will not be updating retroarch on the p/m image as I no longer support that unit.  Wummle's update will be your best option for an updated retroarch for this device.

closed 2021-08-20 by christianhaitian

Turro75, 2021-08-20:
Many thanks Christian, may I ask You which are the technical reasons that made You for end support? I won't ask You for restoring support, it's just to know in case I decide to fork and keep arkos for p alive, I would like to know what are the issues I would have to manage.

christianhaitian, 2021-08-20:
Because of the bad USB controller interface on this unit.  It tends to fail for no particular reason at any moment during game play and it doesn't survive sleep for standalone emulators and ports.

https://github.com/christianhaitian/arkos/wiki/Announcement#442021

https://github.com/christianhaitian/arkos/wiki/Known-issues