christianhaitian/arkos #409: Pico-8

yerrowretro, 2021-08-23:
Can't Install Pico-8
Followed the guide to drop in the 3 files in the pico-8 folder from the EASYROM partition.
There is also a pico-8 folder in Bios that I have not touched.
Pico-8 is not showing up in either ports or emulators.

Thanks

andybarr-Ecolab, 2021-08-23:
Can you see it in the Arkos start menu?

Main Menu (start)> UI settings > Visible systems, then tick the Pico-8 system. I can remember if you have to restart the device.

Andy

yerrowretro, 2021-08-23:
> Can you see it in the Arkos start menu?
> 
> Main Menu (start)> UI settings > Visible systems, then tick the Pico-8 system. I can remember if you have to restart the device.
> 
> Andy

Hi Andy!
Yes... I check and rechecked after a restart and it is not even showing up as an option.

=(

christianhaitian, 2021-08-23:
Did you add any actual pico-8 games to the pico-8/carts folder?

christianhaitian, 2021-08-23:
Taken from here: https://retrogamecorps.com/2020/11/12/guide-pico-8-on-retro-handhelds/#ArkOS

Within the EASYROMS/pico-8 folder, make another folder named “carts”. Inside this folder you can put your own game carts (in .p8 or .png format) and they will appear in the ArkOS main menu.

christianhaitian, 2021-08-23:
Also stated here https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#pico-8

andybarr-Ecolab, 2021-08-23:
I think you can also create an empty text file called zzzsplore.p8 (placed in the carts folder), that will then allow you to connect to the pico-8 splore catalogue

andybarr-Ecolab, 2021-08-23:
also youtube guide here https://www.youtube.com/watch?v=BxED85l9wvo&t=445s

yerrowretro, 2021-08-23:
SOLVED!

Here are a few things.
I did have the "carts" folder under the pico-8 folder, the pico-8 folder is under the root of EASYROM.
I did create a text file called zzzsplore.p8 but it didn't change the extension... it ended up being zzzsplore.p8.txt, changed that.
I have also download a .p8.png and put that in the cart folder as well.
After that Pico-8 is showing up along all the other emulators, not on the Port Emulator.

closed 2021-08-23 by yerrowretro