christianhaitian/arkos #412: Option to turn off Emulation Station boot logo

dabbler77, 2021-09-09:
Just a small suggestion, it would be handy to have a menu option to disable the Emulation Station boot logo (esp for those who are  fully customising the OS)

Adding <bool name="SplashScreen" value="false" /> to es_settings.cfg doesn't seem to work and is restored to defaults

Maybe adding --no-splash to the launch script would work (alas i dont know which script ArkOS uses to launch emulation station!??)

christianhaitian, 2021-09-09:
/usr/bin/emulationstation/emulationstation.sh

christianhaitian, 2021-09-09:
I thought about including this as an option in the passed but then as it boots, it just shows that it's scanning directories for new games and just looks strange without an image.  Better option would be to replace the image with something else during load.  The image is a .svg and is located here:  /usr/bin/emulationstation/resources/splash.svg.  If you come up with something better than that default image, let me know and if ok with you, I'd probably like to include it over the default ES one.

closed 2021-09-09 by christianhaitian

dabbler77, 2021-09-10:
Strange, passing --no-splash dosnt seem to have any affect on the Emulation Station boot screen, have you managed to get this parameter to work in testing?

Removing splash.svg does work, but you are still left with the "directory scanning" at start up.

I could swear that --no-splash used to remove the startup logo and the scanning bar etc etc!