christianhaitian/arkos #413: Support for RG351MP?

markb44z, 2021-09-16:
Hi, any plans to support the newly released RG351MP? Thanks!

christianhaitian, 2021-09-16:
I'm supposed to be receiving a unit soon.  Provided the hardware is good and Anbernic makes sources available, I may release ArkOS for it.  Just waiting until I get my unit so I can further review possibilities.

closed 2021-09-16 by christianhaitian

christianhaitian, 2021-10-01:
ArkOS 2.0 Test Release 1.0 for the Anbernic RG351MP is now available

Google Drive: https://drive.google.com/file/d/1iRGotglxeqiAAUChE_lIvWmwvhUeh0mt/view?usp=sharing
Mega: https://mega.nz/file/nFJgmLgA#il82b4JQOUaqQCLZcVkZ3ZUZoG_Na2lnvdOLuUOeaJI
FAQ: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351MP
Changelog: https://github.com/christianhaitian/arkos/raw/main/changelogs/rg351mp-changelog

**Important Notes**
- Only authentic Sandisk and Samsung cards are supported and recommended.  
  - In my initial testing of my 2 units, 1 unit failed to boot with a PNY sd card flashed with ArkOS.  The same unit would occassionally experience corruption with a Gigastone sd card.  The other unit seemed to boot the same PNY card without issue.  Didn't test Gigastone on it.  **TO MINIMIZE ON POTENTIAL FRUSTRATION, ONLY USE AN AUTHENTIC SAMSUNG OR SANDISK SD CARD**
- Follow the instructions from the wiki to properly loading this image: https://github.com/christianhaitian/arkos/wiki#instructions-for-loading
- Initial boot and preparation can take about 8 minutes to complete due to the themes being copied.  Be patient!
- Be sure during initial boot, **Only have the OS SD card in slot 1 (TF1/INT.) during the initial boot!**.  **Do not put an SD card in slot 2 (TF2/Ext.) during the initial boot!**  Failure to follow this may result in strange behavior during the intial boot preparation process.
- Once the initial boot has completed and you're at the Emulationstation menu, you can insert your second sd card into slot 2 (TF2/Ext.).
- To make use of your second sd card for your games, go to options/Advancec/Switch to SD2 for ROMS.  For more information, see: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg351mp#q-how-do-i-make-use-of-the-second-sd-slot-for-roms-and-games
- There's no boot logo currently due to the unavailability of an updated uboot source code.  Hopefully this will be resolved in the near future.
- Hotkey information: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351MP#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS

Make sure to check for Updates online via the Options section for the latest fixes and features as they become available.  There's already 2 updates already.

markb44z, 2021-10-01:
Excellent job getting it out so quickly, everything seems to work except Atari Lynx none of which load regardless of which core and emulator I select......other than that all good and perfect timing as I received my RG351MP yesterday!