christianhaitian/arkos #414: Feature Request: Use 32-Bit GPSP build to enable dynarec

therealteamplayer, 2021-09-16:
The 32-bit ARM version of libretro's gpsp has a dynamic recompiler, the 64-bit version does not. I compiled and tested it briefly and it seems to give great performance benefits. Would it be worth adding this as an option? It would probably help for edge-case games that struggle in mGBA.

The current GPSP build seems to have issues anyway (graphical errors in games such as mario golf, that are not present when building the latest commit)

christianhaitian, 2021-09-17:
Hello,

When it comes to improving emulation performance, that's always my first priority for ArkOS.  Let me know what the make parameters are that you used for the 32-bit version of gpsp and I can include that in the repo and update the 64bit gpsp emulator in the repo as well.

therealteamplayer, 2021-09-17:
I pasted this into the Makefile on the [libretro/gpsp](https://github.com/libretro/gpsp) repo:
```
# Go-Advance
else ifeq ($(platform), goadvance)
	CPUFLAGS := -Ofast -marm -march=armv8-a+crc+simd -mtune=cortex-a35 -mcpu=cortex-a35 -mfpu=neon-fp-armv8 -mfloat-abi=hard -flto -DUSE_RENDER_THREAD -DNO_ASM -DARM_ASM -frename-registers -ftree-vectorize
	CFLAGS   := -DNDEBUG -Ofast -fno-ident
	LDFLAGS  += -Ofast -fno-ident
	CFLAGS  += $(CPUFLAGS) -fpic -fomit-frame-pointer -ffast-math -fno-rtti -fno-exceptions -fno-non-call-exceptions -Wno-psabi -Wno-format
	LDFLAGS += $(CPUFLAGS) -lpthread -Wl,--gc-sections -shared
	TARGET := $(TARGET_NAME)_libretro.so
	fpic := -fPIC
	SHARED := -shared -Wl,--version-script=link.T -Wl,--no-undefined
	CPU_ARCH := arm
	HAVE_DYNAREC = 1
```
...which was modified from the [libretro/mgba](https://github.com/libretro/mgba) build flags for the go advance. I then ran `make -j3 platform="goadvance" CC=arm-linux-gnueabihf-gcc`.
It seems to work well but haven't tested it for long, may need some tweaking, idk. Building for `rpi3` (already in makefile) also seems to work.
Might be worth noting that I've only just submitted an issue to GPSP for one of the games I was hoping to improve performance on (mario golf).

christianhaitian, 2021-09-22:
I've added the 32bit version of gpsp to ArkOS with today's update.  I've also updated the 64bit gpsp core in repository.  You can just update the core in retroarch and you should get it.  It the 32bit version of gpsp seems to perform the best for Final Fantasy VI from my very limited testing.  Thank you suggesting this.

closed 2021-09-22 by christianhaitian